PeterStaev/lego-spikeprime-mindstorms-vscode #1: Command not found

dreamxjei, 2021-05-15:
Hello, thanks for making this. When I try any of the commands in VSCode, such as 'Lego Hub: Upload Program' I get the error: command 'lego-spikeprime-mindstorms-vscode.uploadProgram' not found. Is there something I should install first?

PeterStaev, 2021-05-15:
Hey @dreamxjei , thank you for reporting this. Seems latest VS Code version bumped Electron version and direct communication over serial ports was failing. I just pushed version 1.0.1 that should have this resolved.

PeterStaev, 2021-05-15:
Forgot to say - you don't need any additional software/code installed. Everything needed is bundled with the plugin itself :)

dreamxjei, 2021-05-15:
Hi @PeterStaev , thanks for the update! I reinstalled the newest version (v1.0.1) and reloaded vscode but am still getting the same issue (command not found). Is it possible that it isn't even getting to the serial ports communication step but is unable to locate the command to begin with?

PeterStaev, 2021-05-15:
Hey @dreamxjei , I guess you are using Windows/Linux? I was testing the plugin on Mac and it is working, but now I tried to load it on my Windows machine and it fails :( I have to see how to make it work across platforms. 

Sorry this is my first VS Code plugin so, I'm a bit of a newb on this 😅

dreamxjei, 2021-05-15:
No problem at all, I just got the set so I'm new to this too. I'm using Windows 10 as you said - please let me know if there are any logs or something I could share to help.

closed 2021-05-16 by PeterStaev

PeterStaev, 2021-05-16:
@dreamxjei , can you please try 1.0.2 version that should be available on the marketplace. I think I've included the correct native depdendencies for windows as well.

dreamxjei, 2021-05-17:
Everything works great, including Bluetooth and printing outputs - thanks again!

tmechen, 2021-07-21:
can confirm this works on windows10, but sadly getting the same error with latest version on linux (ubuntu 20.04.1 lts)

commit 0e3a6857 (2021-05-15): chore: Bump electron to 12.x

Fix #1
commit d6125d12 (2021-05-16): feat: Add binding for win32 platform

Fix #1