PeterStaev/lego-spikeprime-mindstorms-vscode #2: How to get a print-command output?

Neko1993, 2021-05-25:
Hello.

I would like to use "print" in my program to track some variables, but in terminal I only get error messages(no standard output). This is possible in the official app. Is it possible in your? I use standard type of uploading.

Thanks.

PeterStaev, 2021-05-25:
Hey @Neko1993 , are you overriding the default print as readme says? Basically no matter if you are using simple or advance you need to have the following if you want the print to output in the VS Code console:
```python
from util.print_override import spikeprint
print = spikeprint
```

Neko1993, 2021-05-25:
Wow, thanks a lot! I did not notice.

Do you know, is there a command to terminate the execution of the program?

PeterStaev, 2021-05-25:
Yes, just do a an F1 in VSCode and search for lego hub. There should be a `Terminate Running Program` command :)

Neko1993, 2021-05-25:
Yes, but how I can do it programmatically?

After all lines have been executed, the program does not terminate on hub. Usually I put a non-existent command so that the program would exit by mistake

PeterStaev, 2021-05-25:
Ah, I'm not sure about terminating it programatically. Never needed to do this. But considering that even in the official MINDSTORMS app you need to push a button to terminate the app, I doubt it is possible.

Neko1993, 2021-05-25:
Thank you? for answers and extension!)

closed 2021-05-25 by Neko1993

ws1088, 2021-07-02:
> Ah, I'm not sure about terminating it programatically. Never needed to do this. But considering that even in the official MINDSTORMS app you need to push a button to terminate the app, I doubt it is possible.

Most of the Spike community has been using `raise SystemExit` to force the program to exit

ws1088, 2021-08-04:
Is it possible to add a setting to print() to vscode output without 
```py3
from util.print_override import spikeprint as print
```
I have tested it out by adding a line to print whatever is not JSON parsable to: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/blob/cf79c7851e0fb7e0af8dd437dbf63e9d117a891c/src/rpc.ts#L66

And it seems working. Just want to know whether you are willing to make such change? Maybe via a setting?