PeterStaev/lego-spikeprime-mindstorms-vscode #4: Unable to load the extension on Mac OS X, with M1 architecture

rantingdemon, 2021-06-09:
Hello, I was very glad to see this extension!  It would help a lot with writing Python for my new Robot Inventors set.

However, the extension fails to load on my Macbook Air, (M1 ARM64 arch).    This is the output from the console:

> Activating extension 'PeterStaev.lego-spikeprime-mindstorms-vscode' failed: Could not locate the bindings file. Tried:
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/build/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/build/Debug/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/build/Release/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/out/Debug/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/Debug/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/out/Release/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/Release/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/build/default/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/compiled/14.16.0/darwin/arm64/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
>  → /Users/rantingdemon/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.0.2/node_modules/@serialport/bindings/lib/binding/node-v87-darwin-arm64/bindings.node.
> 

I've tried googling around on how to solve this, but I'm at a bit of a loss - I have never tried or needed to use either NodeJs or to build a VS extension.  If you could help me with some tips I would be happy to try and fix it and submit a pull request with my changes.

Thank you!

PeterStaev, 2021-06-09:
Hey @rantingdemon , sadly this is a case of a missing node binding for a platform to which I don't have access to 😞 It would be great if you can help me add the needed files so the plugin can support M1 Macs. 

In order to do so:
* Fork the repo and create a topic branch. 
* Then clone it locally, change to the topic branch 
* Run `npm install` in the root
* Run `npm run build.binding`. 
* The binding file will be located at `<project root>/node_modules/@serialport/bindings/build/Release/bindings.node`. Create the following directory `<project root>/.bindings/node-v87-darwin-arm64` and copy the file there. 

In order to test everything works, simply open the <project root> directory in VS Code and press F5. This will start another instance of VS Code that should have the plugin loaded and working. If everything looks good, submit a PR and I will see to push a version to the Extension Marketplace.

Let me know if you need any further assistance.

closed 2021-06-13 by PeterStaev

PeterStaev, 2021-06-13:
Hey @rantingdemon , thanks for the PR! The binding is added in version 1.0.3 of the plugin.

rantingdemon, 2021-06-13:
Glad I could help!