PeterStaev/lego-spikeprime-mindstorms-vscode #8: Add support for Linux platform

PeterStaev, 2021-07-21:
Currently because there are no bindings for the linux platform the plugin fails with "Command not found" error. 

cc @tmechen

PeterStaev, 2021-07-21:
Hey @tmechen, 

If I would appreciate if you can help with this, since I'm not much of a Linux guy :) You can check my explanation how to create the required bindings and submit a PR with them here: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/4#issuecomment-857440080 The only change would be that the folder in which you should place the file is `node-v87-linux-x64`

tmechen, 2021-07-21:
sry for not including in my initial comment:  
already tried building it, but got tons of warnings and didnt work first try, 
will look again later when not at work :)

commit 4b7b5c95 (2021-08-29): feat: Add linux x64 support (#15)

Close #8
closed 2021-08-29 by PeterStaev