PeterStaev/lego-spikeprime-mindstorms-vscode #9: Is it possible to add a setting to print() to vscode output without

PeterStaev, 2021-08-04:
Is it possible to add a setting to print() to vscode output without 
```py3
from util.print_override import spikeprint as print
```
I have tested it out by adding a line to print whatever is not JSON parsable to: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/blob/cf79c7851e0fb7e0af8dd437dbf63e9d117a891c/src/rpc.ts#L66

And it seems working. Just want to know whether you are willing to make such change? Maybe via a setting?

_Originally posted by @ws1088 in https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/2#issuecomment-892382286_

PeterStaev, 2021-08-04:
Hey @ws1088 , I'm fully for supporting using the "native" `print` command that comes with Python, but when I tried it initially I wasn't getting any input from the JSON RPC. Not sure if they changed something in the brick, but I will experiment and if it possible will see to add support for this.

ws1088, 2021-08-04:
thanks. attached is what I did. line 44's change to replace \n with \n\r like the rest

[image]

commit 34dd02a1 (2021-08-22): feat: Make standard print command output in log

Close #9
closed 2021-08-22 by PeterStaev