PeterStaev/lego-spikeprime-mindstorms-vscode #12: Extension fails to activate

balva, 2021-08-07:
Can repro on two systems where the extension has worked before. May be related to the VS Code upgrade.

Can repro with both v.1.1.0 and v.1.0.3.

macOS 11.5.1 (20G80)

VS Code
Version: 1.59.0
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:14:40.191Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Darwin x64 20.5.0

[2021-08-06 22:02:26.795] [exthost] [info] ExtensionService#_doActivateExtension PeterStaev.lego-spikeprime-mindstorms-vscode {"startup":false,"extensionId":{"value":"PeterStaev.lego-spikeprime-mindstorms-vscode","_lower":"peterstaev.lego-spikeprime-mindstorms-vscode"},"activationEvent":"*"}
[2021-08-06 22:02:26.795] [exthost] [info] ExtensionService#loadCommonJSModule file:///Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/out/extension.js
[2021-08-06 22:02:26.806] [exthost] [error] Activating extension PeterStaev.lego-spikeprime-mindstorms-vscode failed due to an error:
[2021-08-06 22:02:26.806] [exthost] [error] Error: Could not locate the bindings file. Tried:
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/build/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/compiled/14.16.0/darwin/x64/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/lib/binding/node-v89-darwin-x64/bindings.node
    at bindings (/Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/lib/darwin.js:2:36)
    at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1315)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
    at Module.load (internal/modules/cjs/loader.js:982:32)
    at Module._load (internal/modules/cjs/loader.js:823:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)
    at Function.P._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)
    at Function.l._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)
    at Module.require (internal/modules/cjs/loader.js:1006:19)
    at Module.require (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/microsoft-authentication/dist/extension.js:1:38760)
    at v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:698)
    at Object.<anonymous> (/Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/@serialport/bindings/lib/index.js:10:22)
    at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1315)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
    at Module.load (internal/modules/cjs/loader.js:982:32)
    at Module._load (internal/modules/cjs/loader.js:823:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)
    at Function.P._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)
    at Function.l._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)
    at Module.require (internal/modules/cjs/loader.js:1006:19)
    at Module.require (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/microsoft-authentication/dist/extension.js:1:38760)
    at v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:698)
    at Object.<anonymous> (/Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/node_modules/serialport/lib/index.js:2:17)
    at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1315)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
    at Module.load (internal/modules/cjs/loader.js:982:32)
    at Module._load (internal/modules/cjs/loader.js:823:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)
    at Function.P._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)
    at Function.l._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)
    at Module.require (internal/modules/cjs/loader.js:1006:19)
    at Module.require (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/microsoft-authentication/dist/extension.js:1:38760)
    at v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:698)
    at Object.<anonymous> (/Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.1.0/out/extension.js:25:20)
    at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1315)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
    at Module.load (internal/modules/cjs/loader.js:982:32)
    at Module._load (internal/modules/cjs/loader.js:823:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)
    at Function.P._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)
    at Function.l._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)
    at Module.require (internal/modules/cjs/loader.js:1006:19)
    at Module.require (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/microsoft-authentication/dist/extension.js:1:38760)
    at require (internal/modules/cjs/helpers.js:88:18)
[... cut ...]
    at v._doActivateExtension (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:12603)
    at v._activateExtension (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:11704)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async Promise.all (index 0)
    at async v.$activate (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:20144)

balva, 2021-08-08:
Confirming that the extension works on the previous version of VS Code
1.58.2 (Universal)
Commit: c3f126316369cd610563c75b1b1725e0679adfb3
Date: 2021-07-14T22:09:06.581Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Darwin x64 20.6.0

PeterStaev, 2021-08-09:
Hey @balva , 

Because of the way the native node bindings in vscode/electron work, when the vscode version starts using a new electron build there is always a chance the extension will break :( 

## Platforms with added bindings

- [x] macOS x64
- [x] Windows x64
- [x] Windows x86

I've pushed version 1.1.1 which should work just fine on those platforms with the new version of vscode. 

## Platforms with missing bindings (extension will fail to activate):

- [ ] macOS ARM (M1)

If anyone can help generate bindings for that platform and submit a PR that would be awesome 😉

ws1088, 2021-08-10:
For my education, would moving to the latest electron build cause the extension to not work if someone is using an older version of vscode?

PeterStaev, 2021-08-10:
@ws1088 , the extension will work with both new and old version of vscode.

balva, 2021-08-14:
@PeterStaev confirming the fix. Much appreciate all your great work in this space.

closed 2021-08-14 by balva