PeterStaev/lego-spikeprime-mindstorms-vscode #13: Ability to upload multiple/all project files in one go

balva, 2021-08-20:
Essential for working with larger code bases. This would be a major differentiation from the stock Prime App workflow. Inspired by the ev3dev extension for the last gen EV3 bricks (https://marketplace.visualstudio.com/items?itemName=ev3dev.ev3dev-browser)

PeterStaev, 2021-08-20:
Hey @balva , that would be a good improvement, but I'm not sure if the new bricks support multiple files in a project 😞

azzieg, 2022-10-02:
For what it is worth, programs sent to my hub now include a filename :-)

```
{
  "i":"RNd-",
  "m":"start_write_program",
  "p":{
    "meta":{
      "created":1663500597447,
      "modified":1664662362313,
      "project_id":"8e6_GMi87A-_",
      "name":"UHJvamVrdCA0",
      "type":"scratch"
    },
    "size":406,
    "slotid":0,
    "filename":"__init__.mpy"
  }
}
```

PR #58 (2023-09-24): Multi file support/branches
Our first pull request ;)

We added the possibility of splitting code into multiple files.
Imports of the type 'from file import *' will be inlined (like #include in C++ works).
For debugging reasons, the resulting code can (depending on the settings) be saved in the current directory otherwise in the temp directory. 

Fix #13
closed 2023-09-24 by PeterStaev