PeterStaev/lego-spikeprime-mindstorms-vscode #14: Python Advanced detect Exit?

Vinz1911, 2021-08-20:
Hello 👋

This is not an issue but I have a question about the EventLoop. I created a library to handle the Lego Handset and currently I rework it so that it works with the latest version of Micropython for Spike and Inventor Hub. So do you know if it's possible to detect if a stack is stopped to run some cleanup code? I saw there are some events but I didn't saw something for end of a program

thank you and best regards

PeterStaev, 2021-08-22:
Hey @Vinz1911 , if you have 
```python
raise SystemExit
```
in your code this raises a `runtime_error` event:
https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/blob/17552ea21890529fe87c338c5f5b16d6a13298b5/src/rpc.ts#L58-L61
So I guess you can do a cleanup if you receive that event.

Vinz1911, 2021-08-22:
Thank you for your reply, do you have any Idea how can I listen for such events on the Hub?

PeterStaev, 2021-08-22:
Take a look at `src/rpc.ts` in this repo. Basically the brick has a JSON RPC server running and you receive/send commands to it over the COM port to which it is connected.

closed 2021-08-31 by Vinz1911