PeterStaev/lego-spikeprime-mindstorms-vscode #16: Several issues with bare print() feature

ws1088, 2021-08-31:
First of all, thanks for adding the print() without importing spikeprint(). Following illustrate a few issues:

```py
# LEGO type:standard slot:18 autostart
from util.print_override import spikeprint 

for i in range(10):
    print('A: bare hello', i)
    
for i in range(10):
    spikeprint('B: spike hello', i)

for i in range(10):
    print('C: bare hello no end', i, end='')
    
for i in range(10):
    spikeprint('D: spike hello no end', i, end='')

raise SystemExit
```

1. Bare print have extra newlines (compare case `A` with case `B`)
2. Bare print with `end=''` has extra RPC messages (I am not sure if it can be fixed)
3. Spike print with `end=''` overwrites the current line. This I have observed before the addition of bare print(). After some debugging, it has to do with `\r` vs `\n`, IMO.

PeterStaev, 2021-08-31:
@ws1088 , I've noticed the different behaviors, but sadly not much can be done since there are cases where I get the value with a new line inside (not sure why) and in some cases there is no ending new line. That's why I decided to leave the extra new line just so it does not end on the same line as the previous print.

ws1088, 2021-09-16:
@PeterStaev It's most likely because of how we are splitting the lines while reading the serial communications:
```py
        this._parser = this._serialPort.pipe(new SerialPort.parsers.Readline({ delimiter: "\r" }));
```
imagine we are parsing the content in the serial buffer:
```
line_1\r\nline2\r\nline3\r\n
```
and if we split the above by `\r` then will will get the following
```
line_1
\nline2
\nline3
\n   <- this one might be a valid JSON so we will not see this?
```
we should be splitting by `\n` and removing trailing `\r` (or splitted by `\r\n` if possible), so that we will get:
```
line_1
line_2
line_3
```
and add `'\r\n'` to each.

if we want to handle `print('hello', ends='')` we might need to do something more...

PeterStaev, 2021-09-16:
@ws1088 I dont think `\n` is an option. It might fix the "simple" print option but from what i have seen JSON from the RPC are separated by `\r` so I prefer to have correct working JSON RPC than to make "simple" print not print empty lines :)

ws1088, 2021-09-16:
@PeterStaev If my theory is right (haven't debug into it), that `\n` is a valid JSON, we can make sure `data` have more than 3 characters and the first non-space character is `{` (see #19). That way last `\n` can be printed.

PeterStaev, 2021-09-16:
`\n` is not a valid JSON:
[image]

ws1088, 2021-09-17:
@PeterStaev ok, how about this?
```ts
        this._parser.on("data", (data: string) => {
            let json: { [key: string]: any };

            try {
                json = JSON.parse(data);
            }
            catch (e) {
                // When data cannot be JSON parsed we re probably getting text from user's `print` command so we log it
                if (data.length > 0 && data.charAt(0) == '\n') {
                    logger?.info(data.substring(1).replace(/\n/gi, "\n\r"));
                } else {
                    logger?.info(data.replace(/\n/gi, "\n\r"));                
                }
                logger?.info("\n\r");
                return;
            }
```

PeterStaev, 2021-09-17:
@ws1088 I will review this and the other print problem (#19), but for the time being I consider those a nice to have and i want to avoid "hacks" that might break the main JSON RPC communication. In the end if users want to have better printing they can always use the LEGO provided one which shouldn't have such problems.

commit 28af3e15 (2021-09-25): feat: Enhance printing

Close #16
closed 2021-09-25 by PeterStaev