PeterStaev/lego-spikeprime-mindstorms-vscode #17: Activation error on Apple silicon (?)

balva, 2021-09-06:
VS Code Version: 1.60.0
Commit: e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff
Date: 2021-09-01T10:56:36.444Z (M1 build vs 2021-09-01T10:54:53.442Z Intel build ??)
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Darwin arm64 20.6.0

$onExtensionActivationError
mainThreadExtensionService.ts:95

workbench.desktop.main.js:sourcemap:1992 Activating extension 'PeterStaev.lego-spikeprime-mindstorms-vscode' failed: Could not locate the bindings file. Tried:
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/build/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/compiled/14.16.0/darwin/arm64/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/alex/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-1.2.1/node_modules/@serialport/bindings/lib/binding/node-v89-darwin-arm64/bindings.node.

PeterStaev, 2021-09-06:
@balva , this is because I don't have access to Apple Silicon and I can't rebuild the required bindings for the new version of VSCode. Would really appreciate and follow https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/4#issuecomment-857440080 to buid the bindings and submit a PR with them. Note that the bindings should be put in the `.binfings/node-v89-darwin-arm64` folder as this is what the new VSCode will use.

balva, 2021-09-06:
PR is up. Thanks for all your fantastic work on the extension, Peter.

commit ef21d60d (2021-09-08): Updated serial port bindings to to support Apple Silicon (#18)

* updated bindings.node for darwin arm64

* Revert "updated bindings.node for darwin arm64"

This reverts commit e7fad08774b689bba33e09eabfe1be74542be1dd.

* added .bindings/node-v89-darwin-arm64/bindings.node

Fix #17
closed 2021-09-08 by PeterStaev