PeterStaev/lego-spikeprime-mindstorms-vscode #27: Connecting with bluetooth only succesfull on 3rd attempt.

jeromba6, 2022-01-09:
I have the mindstorms Robot Inventor (51515). When connecting to the brick via USB cable first attempt is always successful. But with bluetooth, which I prefer not having any cables in the way, I first have to set the bluetooth connection to serial.
After this startup VSCode try to connect (it fails)
Close VSCode
startup VSCode try to connect (it fails again)
Close VSCode
startup VSCode try to connect (succes :+1: )

This is aways the case. 
My computer is running on Ubuntu 21.10, for now it's my only way to connect to the Lego brick from Ubuntu so I'm very happy to use the plugin.

PeterStaev, 2022-01-11:
Hey @jeromba6 , sadly I'm not a linux user myself (I wasn't even able to compile the plugin on a Ubuntu VM...). Hopefully a more experienced linux user/dev could help on this.

dmusican, 2022-06-16:
I'm a Linux user (Ubuntu 20.04) with a Spike Prime, and I landed here trying to make a bluetooth connection work. I couldn't get it to work at all until I saw the hint from @jeromba6 about setting the bluetooth connection to serial. I don't know if this is what that meant, but after some searching, here's what I came up with. I'm adding it here to help others, but also because if it so happens that this is a different approach, maybe it would help with the reported bug?

- Run the command `bluetooth devices` to see the list of paired devices
- Then run `sudo rfcomm bind 0 <dev>`, where `<dev>` is the hex device id associated with the robot from the above list

That worked. I was then able to connect with /dev/rfcomm0 with VSCode.

@jeromba6: I don't know if I've provided useful info, or just told you what you already knew. At the very least, thanks for getting me started. And further thanks to @PeterStaev for this fantastic VSCode extension.

PeterStaev, 2022-06-16:
@dmusican thanks for the input on this issue. Just to add to this - the plugin communicates over COM with the Brick and not directly on BLE. This is done automatically for Windows and Mac and for both USB and BLE connections (i.e. when you connect the brick to the computer it automatically opens/creates the COM port). I guess on Linux it is not done automatically :(

TheWendyPower, 2022-08-14:
@dmusican I can't get this to work. I could be missing additional drivers. I follow this: https://gist.github.com/0/c73e2557d875446b9603 I can connect to the hub, but I'm unable to complete the `sudo rfcomm bind 0 <dev>` step, as it won't stay connected. It pairs and then disconnects.

dmusican, 2022-08-15:
@TheWendyPower, I think that's by design. It pairs, then disconnects because you're not actually using it. It will then connect again when you actually connect to it from VSCode.

foton, 2023-07-05:
Exactly. I start the BT on Ubuntu and  RI (51515) hub and hub was after while displayed as "LEGO Hub  A8:E2:C1:9C:67:16 disconnected" . According instructions from [lego-hub-tk](https://github.com/smr99/lego-hub-tk) , I used  command`sudo rfcomm connect /dev/rfcomm0 A8:E2:C1:9C:67:16` . 
```
> sudo rfcomm connect /dev/rfcomm0 A8:E2:C1:9C:67:16
Connected /dev/rfcomm0 to A8:E2:C1:9C:67:16 on channel 1
Press CTRL-C for hangup
```
And bluetoothe state in "explorer" changed  to "connected".

I can than select `/dev/rfcomm0` from extension menu.... but it will throw VScode error `Connecting to Hub Failed! Error: Operace zamítnuta, cannot open /dev/rfcomm0`  ( "Operace zamítnuta" is in czech language, translated to english it is "Operation denied/forbidden") .

PeterStaev, 2023-07-29:
>When I connect hub by USB, I can run screen /dev/ttyACM0 115200 and it returns some number, which are changing when I move the hub. But VScode still says, that Connecting to Hub Failed! Error: Zařízení nebo zdroj jsou používány, cannot open /dev/ttyACM0 ("Zařízení nebo zdroj jsou používány" => "device or source is used")
@foton this is a normal error since the COM ports do not allow simultanious connections. So while you have your screen running or you have connected through the LEGO app, no further connections are allowed. You need to close all and then try to connect through VSCode. Hopefully this helps.

PeterStaev, 2025-05-12:
With the plugin rewritten to support HubOS3 which uses direct bluetooth connection, I think this issue is no longer relevant.

closed 2025-05-12 by PeterStaev