PeterStaev/lego-spikeprime-mindstorms-vscode #29: Is it possible to use autocomplete for the spike libraries?

TravelTrader, 2022-03-06:
Thanks for this awesome tool.
I recognized it for connecting the Remote Control to the Hub.

My question is just:
**Is it possible to use autocomplete for the spike libraries?**

Thank you.

PeterStaev, 2022-03-07:
Hey @TravelTrader , this is on my wish list too, but so far did not find a good way to achieve this...

TravelTrader, 2022-03-07:
Can I help you in any way?
I'm new with VSCode but I am coding for many years now. Maybe we can add this feature together in any way?

PeterStaev, 2022-03-11:
Well, this is my first vscode extension as well 😄 Problem is I haven't found a way how to include custom language autocompletions in a plugin. Right now from what I understand they fully rely on the native language service that comes with the given language.

TravelTrader, 2022-03-12:
Respect for this useful extension if it's your first.
But I see the problem with the autocompletions. I'll tell you if I'd find anything, of course.

PeterStaev, 2022-04-10:
Seems it might be possible to use `python.autoComplete.extraPaths`: https://code.visualstudio.com/docs/python/editing#_enable-intellisense-for-custom-package-locations

Not sure if it will be possible to create and distribute mock PY files with all the methods provided from the LEGO hub in the extension and then users to add the extraPaths settings to their workspace and point it to the directory within the plugin. Or even better - to have the extension auto append its autocomplete folder to the list when it is loaded.

Neko1993, 2022-05-30:
Hi!

We use an additional folder in the project directory. It describes all the classes and auto-completion works great. You can also describe the work of functions in your language there.

BlockOG, 2022-06-25:
For the spike library you can just put this file in the same folder as the program you're doing rn https://github.com/sanjayseshan/spikeprime-tools/blob/master/spiketools/hub/spike.py (I'm not sure if it has everything, but probably not since there is spike.control which can't be in the same file)

GSG-Robots, 2023-12-20:
> For the spike library you can just put this file in the same folder as the program you're doing rn https://github.com/sanjayseshan/spikeprime-tools/blob/master/spiketools/hub/spike.py (I'm not sure if it has everything, but probably not since there is spike.control which can't be in the same file)

We use this for us and it does have all functions, just mixed together into one single file. We had to sort out control functions and move them into a seperate file, etc.

Ruimtewese, 2024-01-01:
@GSG-Robots and @PeterStaev the suggested work around using the 'spike.py' file also works with a 'mindstorms.py' file in the root directory.

blake437, 2025-08-01:
Maybe add a file that represents the modules?