PeterStaev/lego-spikeprime-mindstorms-vscode #33: Beginner quick start?

nikobonomi, 2022-04-04:
Hi Guys,

Sorry I'm a total noob with visual studio.

I've installed the plugin on a fresh visual studio install, under ubuntu 21.10.

Now I'm pretty lost, I dont have the icon "hub disconnected" in the bottom status bar nor I know how to start a "spike project". I have donw a quick google search but with no lucky.

I went trough the README.md multiple times but still I have no clue where to start :/

The hub is connected trough an USB cable to the laptop, I'm able to use the lego spike web application to run some code, I also used the screen command to interact with the python on /dev/ttyACM0 device .

Thanks in advice!

[image]

PeterStaev, 2022-04-04:
Hey @nikobonomi , if you dont see the item in the status bar this means that the plugin fails to start. Right now seems the new VSCode version uses a new V8 engine which requires the plugin to be rebuilt (#32), which I will try to do later in the week. But also before I had problems compiling the plugin for Linux, as I'm not a Linux expert myself :( I will try again when I rebuild the plugin for Win/macOS, hopefully it would work this time.

nikobonomi, 2022-04-04:
Hi @PeterStaev!

Thanks for the answer, actually I have tried following the building instruction.

During the build it generates some warning
s and notes, noting special to be honest (I upload the console log), I appears that it is not able to compile the bindings.

Maybe it could be useful to someone familiar with node :)

[logs.txt](https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/files/8409705/logs.txt)

By the way, thanks for you time!

closed 2022-04-10 by PeterStaev