PeterStaev/lego-spikeprime-mindstorms-vscode #35: Cannot connect Lego Hub

kaalen, 2022-04-12:
This extension doesn't seem to work for me. Is it supposed to be "plug and play"?
I don't see any connection status in the bottom status bar.
I tried both USB cable and Bluetooth.
If I try to activate "Lego Hub: Connect to hub" via Command Palette, I get the following error
[image]

PeterStaev, 2022-04-12:
@kaalen make sure you are using the latest version (1.6) of the extension and also latest version of vscode (1.66). If you still do not see the status bar item, let me know what OS you are using it on.

PeterStaev, 2022-04-12:
Actually I just updated the plugin and it does not load :( Not sure what messed up the version uploaded to the extension gallery. Will see to adjust and push another version when possible.

PeterStaev, 2022-04-12:
@kaalen please try version 1.6.1 of the extension. It should resolve the load problem.

closed 2022-04-12 by PeterStaev

kaalen, 2022-04-13:
Confirming it now works. Great work, thanks for your effort. Coding in VSCode is much nicer than the LEGO app.

commit d50606cf (2022-04-12): fix: Add prebuilds to package

Close #35