PeterStaev/lego-spikeprime-mindstorms-vscode #36: Can't exit program after uploading and starting

ADScomics, 2022-04-26:
On macOS Monterey 12.2.1. Upon running a program, if I terminate the program, it doesn't appear to exit the program on the hub. Pressing the center button does nothing either, so I need to hold the button down to restart the hub. I have my hub plugged in via USB. Don't know if this could have any bearing on the problem, but I did go back and forth between the Pybricks firmware, and the default firmware (as in, uninstalling and reinstalling). Unsure if that could have affected the brick in any way.

PeterStaev, 2022-04-26:
Hey @ADScomics , when you say "terminate a program" do you mean the command that the plugin provides or you terminate it in the Python code?

PeterStaev, 2022-04-26:
From what I see from your code you have an endless loop on the main thread. This is what stalls the hub and that's why the central button does not end the program. If you try the same code in the Mindstorms/SPIKE app probably the result will be the same 😃

ADScomics, 2022-04-26:
Gah, that was it! Strangely enough, the Mindstorms app didn't have that same issue, although uploading and running the code was finicky a lot, so maybe that had something to do with it. Thanks for your help! I come from object oriented programming in C sharp, so Python is very new to me. 😅

closed 2022-05-18 by PeterStaev