PeterStaev/lego-spikeprime-mindstorms-vscode #37: spike could not be resolved

MrGibbage, 2022-05-17:
First things first: I am a python newbie. I've been programming Java for many years, but this is my first time doing any python. Our FIRST Lego League team just got some Spike Primes and some of the kids want to use python. What's more, some of them are familiar with python and have used VS code, so this is the platform they'd like to use. So I am doing some testing so I can understand and the first thing I am seeing are two errors that imports spike and spike.control could not be resolved. I understand that this can be ignored and in fact, the play button works and I can run my little program on the Spike Prime. But I can't seem to figure out how to use debug. When I try debugging, it halts on the first import and I can't seem to make it continue.

So I have two questions:
1. How should I deal with the two errors? Do I just leave them there and ignore them, and tell the kids that they aren't a problem?
2. And given that I have these two errors, is it possible to debug Spike Prime programs?

Thanks in advance!
Skip
[image]

PeterStaev, 2022-05-17:
Hey @MrGibbage , straight to your questions:
1. Yes, you can ignore the errors. Those are because the spike specific files are not present in the local python install. I'm looking at possibilities to add auto complete (and basically make those errors disappear) #29 , but so far I did not find any good solution to that. 
2. Sadly there is no possibility to debug programs on the hubs.

closed 2022-05-19 by MrGibbage

SechelTov, 2022-07-22:
Hi
Having same errors (Import "spike" could not be resolved) but unable to ignore - getting a run time error 
> ModuleNotFoundError: No module named 'spike'

Running Win10
Please advise on how to install the library
Many thanks in advance

PeterStaev, 2022-07-22:
@SechelTov , you cannot install the library. You should simply ignore the errors in vscode regarding missing libraries.