PeterStaev/lego-spikeprime-mindstorms-vscode #38: How can I see the programs on the hub?

Sailor247, 2022-06-18:
How can I see the programs on the hub and choose between standard / advanced python?
And can I code similar to the python beta in the app?
I am using Linux and it would be cool to be able to code from my pc.
Thanks!

Sailor247, 2022-06-18:
is there a way to control the motors and sensors?

PeterStaev, 2022-06-18:
Hey @Sailor247 , seems on linux there are some caveats. Checkout the helpful comment here: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/27#issuecomment-1157813751. Hopefully it will help you get started (I'm not a linux user myself so can't be of much help)

For your other questions: 
* You can view/start a program on the hub by first connecting to the hub and then use vscode command pallete to start a program. Simply type "lego" and you will see all the commands from the extension. 
* You chose between standard/advanced when you upload the program to the hub. Checkout the structure of each of the types in the readme. Also you can write a "hint" comment at the start of the filethat simplifies upload/type select. Those are explained here: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode#automatic-uploadstart-of-a-python-file
* Basically the standard python program in the extension is the same as coding python in the LEGO's editor.
* You can't control the motors and sensors directly from the extension/vscode. The extension helps you code python programs and upload/start them on the hub.

closed 2022-08-12 by PeterStaev