PeterStaev/lego-spikeprime-mindstorms-vscode #40: How to use `vm.register_on_button`?

BlockOG, 2022-06-24:


BlockOG, 2022-06-25:
I have found that it needs to do this: `vm.register_on_button(some_constant_idk, handler, "left" or "right", "pressed" or "released")` but I don't really understand what handler has to return, it has to return a list of integers, but what integers Idk

PeterStaev, 2022-06-26:
Hey @BlockOG , thanks for your research! From what I tried there is no need the handler to return anything. For example linked a handler with a `print` command and it worked fined. Added a section in the advanced features doc.

closed 2022-06-26 by PeterStaev

BlockOG, 2022-06-26:
When I tried that I got a NoneType is not iterable error
I also did a print in the handler function and I only saw that print after the program endded, so Idk

PeterStaev, 2022-06-26:
It is working fine when I tried it on my end.

BlockOG, 2022-06-30:
@PeterStaev Could you maybe give me a program that I can test on my spike prime to see if it works? (a program which uses vm.register_on_button)

PeterStaev, 2022-06-30:
@BlockOG this is what i did on my end:
```python
# LEGO type:advanced slot:19 autostart
from runtime.virtualmachine import VirtualMachine

async def on_start(vm, stack):
    print("This is the spot of your code")

async def on_press(vm, stack):
    print("Button pressed")

def setup(rpc, system, stop):
    vm = VirtualMachine(rpc, system, stop, "something_unique")
    vm.register_on_start("another_unique_string", on_start)
    vm.register_on_button("another_unique_string1", on_press, "left", "pressed")
    return vm
```

BlockOG, 2022-07-05:
I got it working in the end, problem was that I had a while true loop in on_start since I was creating games

commit 3c97b8fd (2022-06-26): feat: Add doc for button state handler

Close #40