PeterStaev/lego-spikeprime-mindstorms-vscode #46: How to download to the brick?

dphaine, 2022-08-06:
Hi there!

The instructions read, "Once connected you can start a running program by choosing its slot."

However it doesn’t say where or how to choose its slot.  Can you please elaborate?

Thanks in advance!
Philip

PeterStaev, 2022-08-06:
Hey @dphaine , there are two ways to select a slot:
1. When you use the "LEGO Hub: Run program" you will be asked for a slot and also the type of the program you are uploading. This is show on the screenshots just below the text you quoted above. 
2. The other way is by specifying a predefined file header comment as explained here: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode#automatic-uploadstart-of-a-python-file

dphaine, 2022-08-07:
Thank you for your quick response Peter!

The (obvious) bit that we missed was Cmd-Shift-P and then start typing the command.  It might be nice to include this for the noobs. 

Big thanks for creating this tool. The Lego editor is painfully buggy so we look forward to using vscode. 

Philip

PeterStaev, 2022-08-07:
Yes, the Ctrl+Shift+P is one way to run the program. But also you can use the buttons at the top right of the editor:
[image]
Just keep in mind that the first one ("Run Python File") is the default python one that executes locally, which can't work as locally you do not have the hub sepcific libs. Just look for the "Lego Hub:" prefixed ones, those are the ones coming from this extension :)

closed 2022-08-07 by PeterStaev