PeterStaev/lego-spikeprime-mindstorms-vscode #47: Can't Upload program when unplug USB cable

ttn-c4plus, 2022-08-12:
When I plugged USB cable and choose COM. It connected and I can upload program. After I unplug USB cable and plugged again, I can't upload program during Status bar showing it still connected. I have to reconnect COM and upload.
p/s: I know how to set # LEGO type:standard slot:5, and Can I set default COM like that.
thanks for your Extensions

PeterStaev, 2022-08-12:
Hey @ttn-c4plus once you physically disconnect the hub, the connection is lost and you can't simply plug it and continue where you left. You need to disconnect VSCode and reconnect it again. 

And no, you cannot set the port like you set other stuff with a comment header.

slonopotamus, 2023-03-08:
> You need to disconnect VSCode and reconnect it again.

The problem is that you cannot disconnect. Clicking on connection button just does nothing. The only workaround I've found is restarting VSCode.

GSG-Robots, 2023-12-20:
> > You need to disconnect VSCode and reconnect it again.
> 
> The problem is that you cannot disconnect. Clicking on connection button just does nothing. The only workaround I've found is restarting VSCode.

If you press Shift+P, enter "LEGO" and run the command to connect, you are able to reconnect. Still it should be fixed that when it disconnects unexpectedly, it should be displayed as disconnected.
We use keyboard shortcuts for connecting and uploading, so I did not even know there was another way to connect than through the Shift+P thing.

closed 2024-01-27 by PeterStaev

PeterStaev, 2024-01-27:
With version 1.7.1 of the extension I'm automatically monitoring for closing event of the port and I'm updating the status in the status bar respectively if you unplug the hub from the computer.

commit d591cab8 (2024-01-27): feat: Monitor for disconnect

fix #47