PeterStaev/lego-spikeprime-mindstorms-vscode #48: automatic saving in slot

legotronic, 2022-08-24:
Very good AddOn for connecting the SPIKE hardware with VSC. 😎👌

One observation:
I can no longer select in which memory slot the source should be loaded - this happens "automatically" - with an added note in which slot the source has been stored in the program header.

How can I change this back to "manual selection"?

I have the same behavior on three computers.

PeterStaev, 2022-08-24:
Hey @legotronic , if you want manual selection then you have to remove the added comment header at the top of the file.

legotronic, 2022-08-24:
> Hey @legotronic , if you want manual selection then you have to remove the added comment header at the top of the file.

That's not true...? 😆

Ok - we will check this tomorrow.

and thanks for the quick feedback.

closed 2022-10-17 by PeterStaev