PeterStaev/lego-spikeprime-mindstorms-vscode #54: MemoryError when uploading and starting files multiple times

J0J0HA, 2023-02-03:
When I upload and run my program (ca 1000 lines), it works just fine. When i do this again now, it gives me an MemoryError. If I restart the robot it will work again, but only once every restart. Please tell me what additional infos you need.

PeterStaev, 2023-02-07:
Hey @J0J0HA , where do you see this MemoryError? If it is in the console from the Hub, what happens if you perform the same action but using LEGO's app?
Another thing you can try (if you haven't done already) is to enable in your vscode settings to compile programs before upload:
[image]

PeterStaev, 2023-02-08:
If the LEGO app works fine, probably the problem is with the compile since their app by default compiles sources before uploading. But since the error happens in the `programrunner` which is on the hub, this indicates problem on the hub itself. 

As for v2/v3 - the extension works **ONLY** with v2. You can subscribe to #55 for updates on if/when v3 will be supported.

PeterStaev, 2023-09-24:
No further response so closing this one for now. In case you still have problems, please provide more details.

closed 2023-09-24 by PeterStaev

J0J0HA, 2023-09-25:
Sorry for not answering, the issue is resolved. It worked since I activated compile, but I still needed to use some less objects.
It is more likely the problem was on the hub, because it also happened when I ran Programs I downloaded through the SPIKE Prime App manually.

Unfortunately, I don't know exactly anymore how the issue got resolved, but I never got the issue again and I am happy with your extension :)

Thank you for your assitance and sorry for not responding again.