PeterStaev/lego-spikeprime-mindstorms-vscode #55: Support for HubOS 3

Strohgelaender, 2023-02-07:
In December 2022, Lego released a new HubOS and Spike Prime Version v3.

After upgrading to HubOS 3, we were no longer able to connect the brick with VS Code using this extension. Reverting back to HubOS 2 resolved the issue.

Will there be support for HubOS 3+? Or do I need to stay on version 2 in order to continue using VSCode for programming python?

PeterStaev, 2023-02-07:
Hey @Strohgelaender , I haven't updated personally my hub, since it is the Mindstorms one, and don't want to brake it :) But from what I read on the LEGO's site Python is currently not support on v3 and is expected to land in early 2023. I suspect this might be part of the problem. Also I suspect that they changed their API. Since I've used the reverse engineering done by other on the v2 API, I'm not sure if/when the extension will support v3. 

In short - for now your best bet is to stick with v2.

ffernn-dev, 2023-05-04:
Anyone know if there's a way to downgrade a Spike Prime to V2?

MrGibbage, 2023-05-13:
The newest v3.2.4 Spike App (hub OS 1.2.107) has python (you have to go into the settings to enable python). It's my understanding the 3.2.4 uses bluetooth low energy, so I don't know how to get the windows COM port number since they aren't displayed the same way regular bluetooth devices are. I think if we can crack that nut, then we will be in business again.

MrGibbage, 2023-05-13:
> Anyone know if there's a way to downgrade a Spike Prime to V2?

@ffernn-dev 
If using windows, it seems the only way is to use pybricks:
https://www.youtube.com/watch?v=3pm6iOUBn8E

ffernn-dev, 2023-05-13:
> The newest v3.2.4 Spike App (hub OS 1.2.107) has python (you have to go into the settings to enable python).
Yeah I saw this, but a lot of the functions seem to have been removed or renamed. Any unofficial documentation I found for version 2 doesn't work anymore haha

MrGibbage, 2023-05-14:
True. I have also seen this. There are some interesting new functions too. And the documentation has not caught up yet. I wonder if it is still a work in progress???

MrGibbage, 2023-05-15:
If this is going is going to take a revision to the plugin (pretty likely, I think), I would like to toss in a feature request. This is all based on the user experience with regular BT, so this could all change given the new BTLE usage. For my situation, I coach an FLL team and we have seven laptops and robots for practice. Most of the time a single team member will use the laptop and robot assigned to them. But from time to time a laptop will be used to control a different robot than the one assigned. What that means is the list of BT COM ports gets longer and longer. And with an incoming and outgoing port for each hub, the list is twice as long as it needs to be. It would be great if the list could first of all, only show the outgoing ports because that is the port needed to upload the code. Also, it would be great if the list would show at least the MAC address (prehaps last four) or even better, perhaps show the robot name. Just think about it.
In any case, count me in if you need any help with testing or have questions about how teams are using your code.

PeterStaev, 2023-05-15:
@MrGibbage please post new feature request as new issues. This issue is specifically for HubOS3 support. Added your comments to a separate task, so we will keep any conversation there.

dsplaisted, 2023-08-27:
As of the recently released version 3.3.0 of the Lego Spike App, Python programming support is available again.  Any update on supporting HubOS 3 from the VS Code extension?  It looks like the extension is based on https://github.com/sanjayseshan/spikeprime-tools for connecting to the hub, do you know if anyone has even figured out how to connect to HubOS 3 programmatically?

PeterStaev, 2024-01-27:
For anyone willing to downgrade, seems LEGO provide an online tool to do this: https://spikelegacy.legoeducation.com/hubdowngrade/#step-1

Haven't tried this personally.

PeterStaev, 2024-01-31:
For the record - I've sent a message on X to the LEGO group asking if they are willing to provide API docs for HubOS3 communication so the extension can be updated. So far no response form them yet :(

b3rgman, 2024-01-31:
They don't seem to make any of it public.

PeterStaev, 2024-05-13:
Nice find! I will review and when time permits see to update the extension.

PeterStaev, 2024-10-10:
@GameTec-live , sadly I'm a little short on time and haven't had the time to review the changes and what is needed. Also I do not have any ETA.

GameTec-live, 2024-10-10:
@PeterStaev understandable, but sucks :/

Maybe you have more insight though; did I miss anything or is there really just no way to properly do parallel stuff? Heck, even the docs for the mindstorms micropython api make references to callbacks existing, but no matter what I try, the spike hub wont accept anything :/

PeterStaev, 2025-05-12:
Hey guys, I have some good news - HubOS3 support has been implemented in the v2.0 version of the extension. Currently it supports only Bluetooth connection with plans to add USB too. I've tested on macOS and Windows so linux might have some problems/caveats. 

Happy roboting :)

Strohgelaender, 2025-05-12:
Thanks a lot! I'll tell my team the great news.