PeterStaev/lego-spikeprime-mindstorms-vscode #57: Enhance COM ports lists

PeterStaev, 2023-05-15:
If this is going is going to take a revision to the plugin (pretty likely, I think), I would like to toss in a feature request. This is all based on the user experience with regular BT, so this could all change given the new BTLE usage. For my situation, I coach an FLL team and we have seven laptops and robots for practice. Most of the time a single team member will use the laptop and robot assigned to them. But from time to time a laptop will be used to control a different robot than the one assigned. What that means is the list of BT COM ports gets longer and longer. And with an incoming and outgoing port for each hub, the list is twice as long as it needs to be. It would be great if the list could first of all, only show the outgoing ports because that is the port needed to upload the code. Also, it would be great if the list would show at least the MAC address (prehaps last four) or even better, perhaps show the robot name. Just think about it.
In any case, count me in if you need any help with testing or have questions about how teams are using your code.

_Originally posted by @MrGibbage in https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/55#issuecomment-1547645241_

PeterStaev, 2023-05-15:
@MrGibbage, not sure what do you mean to limit COM ports only to the outgoing. For me It shows just the COM ports on the computer. Also I do not think there is any designation for incoming/outgoing ports. At least the plugin communicates both in and out packets on a single COM port.

MrGibbage, 2023-05-15:
On our WIndows 10 computers, once I have connected a hub to a computer over bluetooth using the Spike app, if you go into the  windows settings, "More Bluetooth Options", you can see the COM ports for each hub. Each hub will have two ports associated with it: an incoming port and an outgoing port. It's the outgoing port that we need in VS code to upload our code. But, when I am in VS code and I click on "Disconnected", we are presented with a list of all COM ports available, including the incoming ports, which can never be used to connect a hub. My suggestion is to not show the incoming ports at all if they cannot be used.

I'll post some screen shots later today.

And thanks @PeterStaev !!!

PeterStaev, 2023-05-15:
Really strange, I'm also using Win10 and I never have seen 2 COM ports for a single hub.

MrGibbage, 2023-05-16:
I have seen this on ten different laptops
[image]
[image]
[image]

Strohgelaender, 2023-06-15:
I have the same issue. When using the extension, I get multiple ports presented where I have no idea which one is the right to choose. I just go through the list until one worked: 
[image]

Thanks to the previous post I also looked into the settings and found incoming and outcoming ports:
[image]

Would it maybe be possible to show the names and incoming / outcoming just like Windows does in this popup? This would eliminate the guessing and allows you to directly choose the right port.

PeterStaev, 2023-07-29:
Sadly, currently I do not see a way to pull the direction of the port nor the name associated.

PeterStaev, 2025-05-18:
With the upcoming version (2.1) that will allow USB connection to HubOS3, the COM list will be filtered and will show only the ports with the correct vendor/product. This should make it easier to select considering there is no way to pull the actual name/direction.

closed 2025-05-18 by PeterStaev