PeterStaev/lego-spikeprime-mindstorms-vscode #63: Connection to Spike Prime HUB fails over USB without error message

UnknownRaisin, 2024-01-11:
Hi all,

sadly I am not able to connect my hub to VS Code, while connecting to Spike App ( Lego Education) works fine.

- HUB: Spike Prime
- Version: 29
- Infos to my OS: Windwos 11, Version 22H2
- VS Code info:
> Version: 1.85.1 (user setup)
> Commit: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
> Date: 2023-12-13T09:49:37.021Z
> Electron: 25.9.7
> ElectronBuildId: 25551756
> Chromium: 114.0.5735.289
> Node.js: 18.15.0
> V8: 11.4.183.29-electron.0
> OS: Windows_NT x64 10.0.22621

Error Message:
[image]

Could anyone help me please to connect the hub device to the VSCode? I didn'nt find any helpfull advices for this case online, sry.

Thank you!

PeterStaev, 2024-01-11:
Hey @UnknownRaisin , can you please open the Output window of VSCode, change from the top right drop down to Extension Host and see if there are any errors regarding the extension. 

Also make sure when connecting you have selected the correct COM port.

UnknownRaisin, 2024-01-11:
Hi @PeterStaev, first of all thanks a lot for your help.
I did what you told me to do.
Here a screenshot:
[image]

Also with the COM5, I just doublechecked, and it seems to be the correct COM....

PeterStaev, 2024-01-11:
@UnknownRaisin what method of connection are you using - USB or Bluetooth?

UnknownRaisin, 2024-01-11:
I tried booth,
while trying to connect via Bluetooth there is following issue: It connects perfectly via the SPIKE IDE. but if I try to connect the HUB directly via Bluetooth, it recognize it but there is a Error Message.
The case I did show you was while using USB, without Bluetooth.

itztantr2, 2024-01-25:
Im having the same issue! Any fixes? @PeterStaev

PeterStaev, 2024-01-25:
Sadly I do not know what might be causing the problem. Also I do not have a Win11 setup on which I can test this. On Win10 I never had problems (if I have selected the right COM port :))

UnknownRaisin, 2024-01-26:
@PeterStaev If you tested it on Win10, and in your documentation you did write that you tested it on MAC too. Have you also tested it on any Linux version? if yes, could you recomend any? or do you have a list? in that case I would like to try it this way. Since I still couldn't make it work on Win 11 (and yes I had 100% the right com ;-) - I suppose it is a Win 11 issue)

PeterStaev, 2024-01-26:
@UnknownRaisin , sadly I'm not a linux guy, so no, can't recommend any. But I've seen on linux people complain about other problems/specifics specifically for bluetooth: https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/issues/27#issuecomment-1157813751

PeterStaev, 2024-01-27:
I've just connected the HubOS to a Win11 VM, and the plugin works fine. By any chance you are running on the new HubOS 3 instead of the legacy HubOS 2?

b3rgman, 2024-01-31:
I'm having issues connecting via USB and Bluetooth as well.  I am on MacOS Sonoma 14.2
USB tells me Resource is busy and Bluetooth fails to connect.
I am on HubOS3

PeterStaev, 2024-01-31:
@b3rgman HubOS3 is not supported currently. Follow #55 for when it will be added.

UnknownRaisin, 2024-01-31:
> I've just connected the HubOS to a Win11 VM, and the plugin works fine. By any chance you are running on the new HubOS 3 instead of the legacy HubOS 2?

Hi @PeterStaev - thats may be possible, I did indeed let the Lego IDE install the newest HUBOS on it do you know where I may get the HUBOS 2? Thanks a lot for testing!

b3rgman, 2024-01-31:
@PeterStaev,
Thank you for the info. I also heard they were going to stop supporting Python but there was a large uproar and they brought it back. In the current Spike 3.4.3, it is supported. 

I am not a fan of downgrading to HubOS 2.  We are actively using these hubs for FLL tournaments. I did see pybricks but I am not sure if that is sanctioned. I'll remove the extension for now and wait for HubOS 3 support.

PeterStaev, 2024-01-31:
@b3rgman , keep in mind that pybricks is also not supported by this extension. You can follow #26 for support for that :)

It would be a shame if they stopped supporting python IMO. Hopefully they never do it.

b3rgman, 2024-01-31:
Understood.  I don't think python is going anywhere with Spike.  I heard starting this year you can now use Python for FRC instead of Java.

itztantr2, 2024-01-31:
> @b3rgman HubOS3 is not supported currently. Follow #55 for when it will be added.

That is why, my hub is V3, my bad

b3rgman, 2024-01-31:
@itztantr2  Im using functions from HubOS3 in my robot game.  I wanted to teach the students Python but want to use VSCode due to not liking the SPIKE interface for such task.  

All we can hope for is that this extension is ported to HubOS3.

itztantr2, 2024-01-31:
Well I just downgrade all my classrooms hubs since it seems spike OS2 is
better for competition due to function such as equilibrium control and
motor differentiation that they remove in OS3 for some reason

Chiraphon Tan Suphagrum

b3rgman, 2024-01-31:
Interesting @itztantr2.  Mine were upgraded automatically when the original SPIKE app was called legacy.  When I am done competing I may look into this approach.

UnknownRaisin, 2024-02-02:
@PeterStaev could it be possible, to get a list of what schould bi installed/ done so that it works fine?
Maybe I'm missing a step. That would be really nice!
Thanks a lot in advance

PeterStaev, 2024-02-02:
@UnknownRaisin there is nothing to be installed on your computer except this extension. But the hub should be running on HubOS2 as v3 is not supported. If you are already on v3 LEGO provides a tool to downgrade to v2 (haven't used it personally): https://spikelegacy.legoeducation.com/hubdowngrade/#step-1

itztantr2, 2024-02-02:
The downgrade solution you provided can no longer be perform on window PC. I suggest doing this instead 
https://youtu.be/3pm6iOUBn8E?si=TMsZ6TUOGNdYtUKi

PeterStaev, 2025-05-11:
Since this issue seems to be related to HubOS3, I will close this in favor of #55 to keep things clean. If anyone has problems with old HubOS2, don't hesitate to open a new issue.

closed 2025-05-11 by PeterStaev