PeterStaev/lego-spikeprime-mindstorms-vscode #66: Support for multiple file programs

JacobenVosloo, 2024-09-22:
Based on this PR https://github.com/PeterStaev/lego-spikeprime-mindstorms-vscode/pull/58

And the read me
[image]

I would assume that splitting code into multiple files work

But based on my simple test it does not

[image]

Can anyone please assist as to what am I doing wrong?
@GreenGitHub99

I also tried changing the settings to precompiled mpy file 

[image]

PeterStaev, 2024-09-22:
@JacobenVosloo , could you please provide the contents of the `main.assembled.py` file (it should be in the same folder as your `main.py` file)?

JacobenVosloo, 2024-09-23:
@PeterStaev, thankyou for the quick response. 

The contents is the same as the main.py file

```
# LEGO type:basic slot:0 autostart

# Use wildcard import for utilities
from utilities.py import *

# Now you can call battery_check directly
battery_check()
```

And the file is located in the same location as main.py and utilities.py

[image]

PeterStaev, 2024-09-23:
@JacobenVosloo I see your import in the assembled file looks different than the one in the `main.py` file - it hast the `.py` extension at the end of utilities. I think this is what causing the problems as the logic automatically adds the extension when it looks up the file to be assembled. Can you check and remove the extension so the import is:
```python
from utilities import *
```

JacobenVosloo, 2024-09-25:
@PeterStaev 

I double-checked, and import is now as you suggested. But it is still not working. Here is a screenshot showing the whole setup and the resultant main. assembled.py file.

[image]

I am working of a Mac, could this be the problem? 

Is there perhaps some example file that I can use to test the setup on my side. 

Thank you again for the support and a great library

PeterStaev, 2025-05-10:
@JacobenVosloo  While working on rewriting the extension with support fro the new HubOS3 I noted that the current prepocessor has a bug so that it "assembles" the files only if the import ends with a space or a tab. Can you try to add one and the end of the import and see if you will get better results?

PeterStaev, 2025-05-18:
Since I fixed the problem with the ending space, I will close this for now. If you experience again the problem with the 2+ version don't hesitate to reopen this one or create a new one.

closed 2025-05-18 by PeterStaev