PeterStaev/lego-spikeprime-mindstorms-vscode #67: Random Exceptions when importing functions

yannerous, 2024-11-23:
When importing functions from a file an exception is sometimes produced by the handle_write_package function without any clear pattern. They all produce the error message:
```
Traceback (most recent call last):
  File "protocol/ujsonrpc.py", line 1, in _handle_message
  File "commands/program_methods.py", line 1, in handle_write_package
Exception:
```
If you need more information from me, don't hesitate to ask.

**My main file:**
```
# LEGO type:advanced slot:0 autostart

from spike import PrimeHub
from spike.control import wait_for_seconds
from runtime.virtualmachine import VirtualMachine # Needed for newer hub versions (3.1.43+)
from test12 import *

async def on_start(vm, stack):
    hub = PrimeHub()
    print("Startup")

    vm.stop()

def setup(rpc, system, stop):
    vm = VirtualMachine(rpc, system, stop, "something_unique")
    vm.register_on_start("another_unique_string", on_start)
    return vm
```

### Examples of files that could not be imported:
**Example 1:**
```
def test():
    return

def test2():
    return
```
**Example 2:**
```
def test():
    #test
    return

```

**Example 3:**
```
#test

def test():
    return

```

### Examples of files that _can_ be uploaded to the spike prime:
**Example 1:**
```
def test():
    return
```
Example 2:
```
#test
def test():
    return

```

yannerous, 2024-11-24:
After further testing, I found that the errors disappear for all examples if I remove the unused import wait_for_seconds or use it in the file: 

```
# LEGO type:advanced slot:0 autostart

from spike import PrimeHub
from spike.control import wait_for_seconds
from runtime.virtualmachine import VirtualMachine # Needed for newer hub versions (3.1.43+)
from test12 import *

async def on_start(vm, stack):
    hub = PrimeHub()
    print("Startup")
    
    #use the imported function in the file
    wait_for_seconds(1)

    vm.stop()

def setup(rpc, system, stop):
    vm = VirtualMachine(rpc, system, stop, "something_unique")
    vm.register_on_start("another_unique_string", on_start)
    return vm
```

I will keep this issue open, in case someone else experiences similar problems or if there is an easy fix available.

PeterStaev, 2025-05-12:
Hey @yannerous , I've rewritten the plugin with support for HubOS3 where the structure of the programs is completely different so I suspect this issue is no longer valid. I will close it but if you experience similar problems with the new version, do not hesitate to open a new issue.

closed 2025-05-12 by PeterStaev