PeterStaev/lego-spikeprime-mindstorms-vscode #68: Get output of program

epicprogram23, 2025-01-19:
This is a question, but i don't know how to add the tag

I can run Print on the hub and it will print it to my computer. is it possible to read the output? The computer knows what it is but how to access it?

PeterStaev, 2025-05-11:
Hey @epicprogram23 , if you mean for another program to be able to access what the hub writes back and appears in the vscode console - I do not think this is possible, and I'm not sure it is in the scope of this extension. Probably best to write your own code that will make the connection to the hub and receive and process the message. 

May I ask what do you want to do with the intercepted printed messages?

epicprogram23, 2025-05-13:
Thanks for responding
I wanted the data it sent back from this extension because it was much faster than other extensions i used         to sense orientation
I'm not working on it right now but i'll remember the advice

closed 2025-05-14 by PeterStaev