PeterStaev/lego-spikeprime-mindstorms-vscode #73: Thanks and some linux install headaches

jvolkening, 2025-06-08:
Many thanks for your time in writing this extension -- I'm planning to use it to help our FLL team learn Python programming on the Spike hub next year.

I installed your extension first on a Windows laptop and it worked great out of the box.

Now I've been working on installing it on a Linux laptop running Debian 12 and VSCode 1.100.3, and the process was not straightforward. I thought I'd document here what I had to do and ask if there is anything to be done in the code base itself to address these issues.

When I first installed this extension in VSCode on Linux, there were no errors and it was listed as installed but none of the additional widgets appeared. The `exthost.log` included (abbreviated):

```
2025-06-08 14:00:51.790 [error] Activating extension PeterStaev.lego-spikeprime-mindstorms-vscode failed due to an error:
2025-06-08 14:00:51.790 [error] Error: Cannot find module '@abandonware/bluetooth-hci-socket'
```

After installing NodeJS/NPM and quite a bit of trial and error which I won't go into here, I was able to get this module installed by switching to the extension install directory and running the following:

```bash
cd $HOME/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-3.0.0
npm install @abandonware/bluetooth-hci-socket
# the following two steps were necessary to avoid NodeJS version mismatch errors
npm install --save-dev @electron/rebuild
node_modules/.bin/electron-rebuild -v 34.5.8
```

After this, everything seems to be working properly. No doubt the electron version number (34.5.8) will change depending on the VSCode version installed and which version of electron is compiled into it.

Is there a reason that `@abandonware/bluetooth-hci-socket` is not included in the installed extension by default?

jvolkening, 2025-06-08:
Completely understandable. Most VSCode users are probably on Windows so the time investment to fix might not be worthwhile. I only installed it myself specifically for using your extension, and our students will all be using Windows.

Feel free to close this issue for now as wont-fix if you want. If I ever have time to work on it myself and make any progress, I'll report back here.

jvolkening, 2025-06-08:
I was able to build your extension from source on Linux into a VSIX file that appears to install and work correctly. However, I have no idea how to adapt the steps I took into whatever deployment workflow you are using, and don't know if this will break anything on other platforms. Steps were as follows (within a fresh checkout of the `master` branch):

Modify `webpack.config.js`:

```diff
diff --git a/webpack.config.js b/webpack.config.js
index 50b849b..b405183 100644
--- a/webpack.config.js
+++ b/webpack.config.js
@@ -9,7 +9,9 @@ const webpack = require("webpack");
 /** @type {import('webpack').Configuration}*/
 const config = {
     target: "node", // vscode extensions run in a Node.js-context 📖 -> https://webpack.js.org/configuration/node/
-
+    node: {
+        __dirname: false,
+    },
     entry: "./src/extension.ts", // the entry point of this extension, 📖 -> https://webpack.js.org/configuration/entry-context/
     output: {
         // the bundle is stored in the 'dist' folder (check package.json), 📖 -> https://webpack.js.org/configuration/output/
@@ -23,12 +25,11 @@ const config = {
         {
             vscode: "commonjs vscode", // the vscode-module is created on-the-fly and must be excluded. Add other modules that cannot be webpack'ed, 📖 -> https://webpack.js.org/configuration/externals/
         },
-        "@abandonware/bluetooth-hci-socket",
         "ws",
     ],
     resolve: {
         // support reading TypeScript and JavaScript files, 📖 -> https://github.com/TypeStrong/ts-loader
-        extensions: [".ts", ".js", ".json"],
+        extensions: [".ts", ".js", ".json", ".node"],
     },
     module: {
         rules: [
@@ -41,6 +42,10 @@ const config = {
                     },
                 ],
             },
+            {
+                test: /\.node$/,
+                loader: "node-loader",
+            },
             {
                 test: /\.json$/,
                 type: "asset/inline",
```

Then,

```bash
npm install -g @vscode/vsce
npm install -save-dev node-loader
npm install --save-dev @electron/builder
./node_modules/.bin/electron-rebuild -v 34.5.8
vsce package
```

PeterStaev, 2025-06-14:
Hey @jvolkening, I've just tried this and here is what happens - when I try to install the HCI socket it complains for an incorrect OS:

[image]

I tried to install it using `--force` but then it fails to load/find it. I suspect that I will have to split my build and do one build for macOS/win and then another build and package for linux. I will have to see how to automate this using GH actions as right now I'm manually building and publishing the package from my dev machine.

PeterStaev, 2025-06-15:
Hey @jvolkening , please try build 3.0.1 - this should include platform specific version for linux that includes the HCI package and should work out of the box on linux systems.

PeterStaev, 2025-06-17:
Well this will be a problem - in general all the other native bindings have been compiled using pre-builds and have been working in the last dozen of version of vscode w/o the need to rebuild for each new vscode version. I will review the HCI code again and see it can somehow be overcome. 

Thank you for providing the feedback!

PeterStaev, 2025-06-21:
Hey @jvolkening , I've just pushed version 3.1.0 in which I've replace the bluetooth package with another fork that seems better supported and provides prebuilds for the HCI socket too. Can you please try and see if that will work better on linux?

jvolkening, 2025-06-23:
Thanks, I appreciate the work you're putting into it. Tried 3.1.0 on Linux and got:

```
2025-06-23 12:37:53.263 [error] Activating extension PeterStaev.lego-spikeprime-mindstorms-vscode failed due to an error:
2025-06-23 12:37:53.263 [error] Error: Module did not self-register: '/home/jeremy/.vscode/extensions/peterstaev.lego-spikeprime-mindstorms-vscode-3.1.0/prebuilds/linux-x64/@stoprocent+noble.glibc.node'.
```

jvolkening, 2025-06-26:
Updated to v3.1.1: no errors this time and the expected LEGO Hub bits show up in the UI. I haven't been able to try actually connecting to a LEGO hub, but as soon as I can I'll return here with an update.

Thanks again!

jvolkening, 2025-08-24:
I was finally able to test this, sort of...everything installs and loads on a fresh VS Code install on Linux with no errors, and I could connect and interact normally with a Spike Prime hub over USB. Unfortunately, I was never able to get the Bluetooth adapter on this laptop working in Linux so cannot test that functionality. The Linux/Bluetooth issue is not specific to this extension -- I couldn't get it working at all in any software -- so I'll close this issue as completed. Thanks again.

closed 2025-08-24 by jvolkening