PeterStaev/lego-spikeprime-mindstorms-vscode #76: Feature request: Real-time sensor feedback in UI

jvolkening, 2025-08-08:
One of our FLL students has been learning the Python API for Spike Prime and has worked in both VS Code using this plugin as well as the LEGO Spike Prime app. I asked her what her preference was so far, and she said the LEGO app because it showed real-time feedback from the hub and sensors when connected.

Now, there are several advantages of using VS Code vs the LEGO app (ability to use version control, ability to modularize code into separate files, etc). However, I see her point that the real-time feedback is extremely useful when they are working on coding mission runs.

It should be possible to add this feature to your plugin as demonstrated in the Python example code in the LEGO repo:

https://github.com/LEGO/spike-prime-docs/blob/446549146df626f5d7f332b0ea3cd0205ce23711/examples/python/app.py#L178C1-L185C24

https://github.com/LEGO/spike-prime-docs/blob/446549146df626f5d7f332b0ea3cd0205ce23711/examples/python/app.py#L117C9-L145C1

Received device messages could be formatted into a compact, single-line format that could be displayed somewhere in the VS Code UI. I'd be willing to help with this, but I currently know nothing about how the plugin interacts with VS Code and my knowledge of JavaScript is rudimentary. Is this something you'd consider implementing at some point?

PeterStaev, 2025-08-09:
Hey @jvolkening , I think this is a great idea! For sure I would consider it, just have to find a good place to put the values in vscode UI :)

jvolkening, 2025-08-24:
I can imagine two possible places to put a UI element for this. The first would be a one-liner in the status bar where the extension has already added the "Unconnected"/"Connected" status/button. Advantages: possibly easier implementation to add on to what the extension is already doing. Disadvantages: limited to single line, might not fit well on smaller screens.

The second option would be to add a tabbed window in the same area where the "Problems", "Output", "Terminal", etc, windows are placed. Advantages: more room to work with, probably more flexible. Distadvantages: not sure if this is even possible for an extension to do.

Here is some working code written using the hub Python API that prints out a one-line status of any/all connected motors/sensors as well as the current yaw from the gyro. This is code that runs on the hub and prints to the terminal, so it can't be used while running/testing other code. It seems it should be possible to translate this into the Javascript communication protocol and add it into this extension somehow.

```python
# LEGO slot:18 autostart

import color_sensor
import device
import distance_sensor
import force_sensor
import motor
import runloop
from hub import motion_sensor
from hub import port

def status_line():
    strings = []
    # iterate over each of the six hub ports A-F
    for i in range(ord('A'), ord('F')+1):
        letter = chr(i)
        port_id = getattr(port, letter)
        status = 'NA'
        dev_id = device.id(port_id)
        if dev_id in [48, 49, 65]:  # motors
            status = f"{motor.absolute_position(port_id)}°"
        elif dev_id == 61: # color sensor
            status = f"{color_sensor.reflection(port_id)}%"
        elif dev_id == 62: # distance sensor
            status = f"{distance_sensor.distance(port_id)}mm"
        elif dev_id == 63: # force sensor
            status = f"{force_sensor.raw(port_id)}f"
        strings.append(f"{letter}:{status}")
    strings.append(f"yaw:{motion_sensor.tilt_angles()[0]}°")
    return ' | '.join(strings)

async def main():
    while True:
        print(status_line())
        await runloop.sleep_ms(500)

runloop.run(main())
```